bambulab/BambuStudio #6230: filament selection

lfigueroa87, 2025-03-27:
### Bambu Studio Version

2.00000

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS sequoia 15.2

### Printer

p1s, a1

### How to reproduce

slice plate
print plate
under filament  - not able to select other colors in my ams 

### Actual results

not able to pick a different color

### Expected results

if my model is red in the slicer but I have blue and pink on my ams . the program will automatically select pick because is close to the color on the slicer. if I would like to select other color in my AMS it won’t let me change.

### Project file & Debug log uploads

[debug_Mon_Mar_24_19_49_03_45683.log.0.zip](https://github.com/user-attachments/files/19483907/debug_Mon_Mar_24_19_49_03_45683.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-27:
Hello, if you have a Windows laptop, you can use the Windows version first to avoid this issue. We have fixed this issue and will release it as soon as possible. so sorry.

If you continue to use a Mac laptop, you can avoid this problem by following these steps：
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally.
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.

[image]

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00