bambulab/BambuStudio #6231: BambuStudio 2.xx BACKWARDS compatibilty lost. WHY!?!?

bolsoncerrado, 2025-03-27:
### Bambu Studio Version

2.xx

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Additional system information

M1Ultra

### Printer

X1, P1

### How to reproduce

1. Upgrade to Bstudio 2.x

### Actual results

2. Try to open any Bstudio 1.x gcode file
3. You get an error and the app closes.

### Expected results

3. You could at least SEND the old 1.x gcodes to PRINT!

WHY NOT!?!?!?!

I can understand Bstudio 1.xx won't open Bstudio 2.xx gcode files but WHY IN THE  NAME OF THE MCMUFFIN Bstudio2.x won't open legacy Bstudio 1.xx gcodes!?!?!!?

For a FARM this is a MESS! We have to re-generate all the gcode files again!?!

Either this is a bug or PLEASE ADD A LEGACY CHECKBOX like Adobe and other companies do when there's a big software change to keep backwards compatibility for a while!!!

### Project file & Debug log uploads

Not needed, really!

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-1, 2025-03-27:
Hello, could you please provide the Gcode.3mf file that caused the import error?

bolsoncerrado, 2025-03-27:
@BoBoYu-1  sure use this one for example:

[BAMBULAB CUTTER BLOCKER P1P.gcode.3mf.zip](https://github.com/user-attachments/files/19485499/BAMBULAB.CUTTER.BLOCKER.P1P.gcode.3mf.zip)

BoBoYu-1, 2025-03-27:
@bolsoncerrado Hello, we tried importing the file you provided on our end and did not encounter the error you mentioned. Could you please let us know if there were any additional steps you took? This will help us better understand and address the issue.
[image]

bolsoncerrado, 2025-03-27:
OK apparently the issue is if you "double click" a BS1.0 gcode.3mf file WITH BSTUDIO 2.0 *CLOSED*.

It works fine if you click OPEN PROJECT and choose the gcode manually with BSTUDIO 2.0 already opened.

bolsoncerrado, 2025-03-27:
https://github.com/user-attachments/assets/42bd4136-80a0-42c1-b478-74e14204ec0b

BoBoYu-1, 2025-03-27:
I followed your steps again and re-performed the operation. However, I still can import it on my side.
[image]

bolsoncerrado, 2025-03-27:
Well you can see my issue exactly on the screen. I don't know what else to say :?

bolsoncerrado, 2025-03-27:
I tried disabling Develop mode and no changes.

SO perhaps you can tell me in WHICH condition did you set the software to display THIS message!??!!

[image]

bolsoncerrado, 2025-03-27:
Well, know what? Let me powercycle (reboot) the Mac. I haven't eversince I installed the new version.... you never know!

bolsoncerrado, 2025-03-27:
No, as you saw in my video, I have no opportunity to interact with BStudio at all. THe popup shows up on the screen and all I can do is clicking OK. If I click OK an empty project shows up.

BTW, just to leave this clear: gcodes created with BStudio 2.0 and DOUBLECLICKED with Bstudio 2 closed OPEN FINE!

bolsoncerrado, 2025-03-27:
Which is the third place btw? I Can only see two marked green?

Haidiye00, 2025-03-27:
Hello, currently there is a dropdown menu for adjusting the machine type, adding or reducing consumables. In the GCode interface, clicking on these three places can trigger this pop-up.

bolsoncerrado, 2025-03-27:
OK I have no explanation to this: If i create a .gcode with BStudio 2, saving it to disk, closing BStudio 2 and then doubleclicking the .gcode.3mf CREATED WITH BSTUDIO2, IT WILL BRING THE SAME ISSUE TOO!!!

Haidiye00, 2025-03-27:
Thank you very much, I have reproduced this issue. Let me analyze it~

bolsoncerrado, 2025-03-27:
WOw! Finally! ;)

Thanks!!

Haidiye00, 2025-03-27:
Hello, I found that you are connected to the machine and triggered this bug. If you log out of the machine connection or software account, there will be no problem during testing. We will fix it as soon as possible~

Haidiye00, 2025-03-27:
[image]
click this，and close your software,then  open your gcode 3mf.this is a bug,We will fix it as soon as possible~

bolsoncerrado, 2025-03-27:
Understood, thanks!

bolsoncerrado, 2025-03-27:
Confirmed.

Logged out.
Closed BStudio 2.
Doubleclicked any .gcode 
Opens fine.
IF I log in without closing BStudio 2, as soon as I log in, the error reappears ;)

Thanks!

Haidiye00, 2025-03-27:
Hello, I apologize for any inconvenience caused. Thank you for your discovery. We will fix it as soon as possible~

closed 2025-04-16 by Haidiye00