bambulab/BambuStudio #6232: Filament calibration from the AMS not possible

McUser95, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.3.2

### Printer

Bambu Lab X1C

### How to reproduce

If you want to do a flow dynamics or flow rate calibration, it is not possible to select a filament from the AMS. Although the AMS is connected. This is only the case after the update to version 2.0.0.95.

### Actual results

No selection of filament from the AMS possible

### Expected results

Selection of filament from the AMS for calibration

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-27:
Thank you for your feedback. We are currently in the process of rectifying it. The next version will be released with the corrections.

sergeantbort, 2025-03-27:
this is occurring on Linux as well.  Please ensure that is also updated

Lawrence42441, 2025-03-27:
Its happening on windows as well. how about you guys start making updates that dont cripple our printers.

Lawrence42441, 2025-03-27:
When exactly will this issue be rectified. this the second time your poor support has caused my studio delays and money.

arsuya3d, 2025-03-27:
I can no longer add filament profiles. I would appreciate it if you could fix this. I received a product for filament promotion, but I can't use it because I can't create a profile. In short, I'm struggling unnecessarily... My device: 10 x X1C :SSSS

cesaratb, 2025-03-27:
Same issue here on my P1S.

xwzkj, 2025-03-28:
same issue on my A1

arsuya3d, 2025-03-28:
The problem has been solved with today's update...

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab