bambulab/BambuStudio #6233: Number Enter Key not working.

KennyGreg, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU-Z TXT Report
-------------------------------------------------------------------------

Binaries
-------------------------------------------------------------------------

CPU-Z version			2.09.0.x64

Processors
-------------------------------------------------------------------------

CPU Groups			1
CPU Group 0			16 threads, mask=0xFFFF

Number of sockets		1
Number of threads		16

Timers
-------------------------------------------------------------------------

	ACPI timer		3.580 MHz
	Perf timer		10.000 MHz
	Sys timer		1.000 KHz

Processors Information
-------------------------------------------------------------------------

Socket 1			ID = 0
	Number of cores		8 (max 8)
	Number of threads	16 (max 16)
	Number of CCDs		1
	Manufacturer		AuthenticAMD
	Name			AMD Ryzen 7 3700X
	Codename		Matisse
	Specification		AMD Ryzen 7 3700X 8-Core Processor             
	Package 		Socket AM4 (1331)
	CPUID			F.1.0
	Extended CPUID		17.71
	Core Stepping		MTS-B0
	Technology		7 nm
	TDP Limit		65.0 Watts
	Tjmax			95.0  C
	Core Speed		4044.3 MHz
	Multiplier x Bus Speed	41.25 x 98.0 MHz
	Base frequency (cores)	98.0 MHz
	Base frequency (mem.)	98.0 MHz
	Instructions sets	MMX (+), SSE, SSE2, SSE3, SSSE3, SSE4.1, SSE4.2, SSE4A, x86-64, AES, AVX, 

### Printer

Bambu Labs P1S

### How to reproduce

1 open bambu studio
2 select project
3 click on resize
4 change size number on x, y, or z using keyboard number pad
5 hit enter on keyboard number pad.
6 nothing happens.  The new numbers don't enter.
7 hit enter on the alphanumeric part of the same keyboard, and the new numbers enter.
8 see picture.

[image]

### Actual results

The number enter key doesn't enter.

### Expected results

With the old version of bambu studio, when i entered new numbers into the x, y, and z areas, i could use the number pad enter key with no problem.

### Project file & Debug log uploads

[Body_mod.zip](https://github.com/user-attachments/files/19487661/Body_mod.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

guanyun-gudujian, 2025-03-28:
fixed 
will be released recently
thanks

mg16373, 2025-03-28:
iMac the same issue. Arrow-Keys & Backspace not working

GuDuJian-J-Zhang, 2025-05-20:
@KennyGreg 
fixed, pls try the latest version, thanks
https://github.com/bambulab/BambuStudio/releases/download/V02.00.03.54/Bambu_Studio_win_public-v02.00.03.54-20250424182834.exe

@mg16373 
pls try the latest version:
https://github.com/bambulab/BambuStudio/releases/download/V02.00.03.54/Bambu_Studio_mac-v02.00.03.54-20250424182834.dmg

guanyun-gudujian, 2025-05-20:
@KennyGreg
fixed, pls try the latest version, thanks
https://github.com/bambulab/BambuStudio/releases/download/V02.00.03.54/Bambu_Studio_win_public-v02.00.03.54-20250424182834.exe

@mg16373
pls try the latest version:
https://github.com/bambulab/BambuStudio/releases/download/V02.00.03.54/Bambu_Studio_mac-v02.00.03.54-20250424182834.dmg