bambulab/BambuStudio #6234: Calibration no longer works in latest version

lulzkiller666, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS  15.3.2

### Additional system information

Mac Mini m4

### Printer

p1s with AMS

### How to reproduce

Go to "Calibrate"
Click "flow rate"
Try and of flow rate tests

### Actual results

Printer stuck in "heating bed" for 5 minutes
Fails with error " Failed to get AMS mapping table; please select "resume" to retry (0700-8012 1665127)

### Expected results

To run a proper flow rate test

### Project file & Debug log uploads

No logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sergeantbort, 2025-03-27:
Experiencing the same issue from Linux as well.  It doesn't appear to be syncing the AMS data from the printer any longer.

I will also note I get the same error now as well when sending the calibration from Orca Slicer as well, this may not be a "slicer" issue but a cloud issue.

cesaratb, 2025-03-27:
Same issue here on my P1S.

FlewByYou, 2025-03-27:
Same issue on X1C on latest firmware.

jrgemignani, 2025-03-27:
Same issue with me #6243

Tobik333, 2025-03-28:
A1 - manual calibration also does not work - for Flow Dynamics and Flow Rate.

Only automatic calibration works for Flow dynamics.

jrgemignani, 2025-03-28:
AMS now shows up in calibration tab on version 2.0.1.50 :) Running a calibration now.

sergeantbort, 2025-03-28:
> AMS now shows up in the calibration tab on version 2.0.1.50 :) Running a calibration now.

They still haven't released the Linux version with the update.. hopefully it's on the way, glad to hear it's been resolved.

jrgemignani, 2025-03-28:
Was able to do both calibrations using the AMS successfully with 2.0.1.50

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab