bambulab/BambuStudio #6235: Version 2.0.0.95 not seeing micro SD card

KennyGreg, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 7 3700X 8 core Processor
32 GB Ram
64 Bitt Windows 11
NVIDIA GeForce RTX 2060 SUPER

### Printer

Bambu Labs P1S

### How to reproduce

Go to "Storage" in Bambu Studio and See the what it says

[image]

### Actual results

The Software wont read the Micro SD card. and it shows "storage icon" as being abnormal.

[image]

[image]

### Expected results

I should be able to see my timelapse and my models.  It works on the old versions of Bambu Studio and also still works on "Orca Slicer"

### Project file & Debug log uploads

[v converter box.zip](https://github.com/user-attachments/files/19493435/v.converter.box.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-28:
Hello, could you please try to remove the SD card from the printer and then reinsert it? Or try using a different SD card? It might be that the SD card is damaged, or the printer failed to recognize it.

KennyGreg, 2025-03-29:
Hello BOBOYU-1
    I’m not sure if you read my entire report, but I did mention that the SD card is seen on the older versions of Bambu Studio, if I uninstall the new version and reinstall  the older versions.  It’s also seen and works correctly if I run “ORCA Slicer” instead of the new Bambu Studio.  The problem appears to be in the version 2 of the Bambu Studio.  It even still happens with the latest update of Bambu Studio.  It’s not my printer or the card.  They both work fine.
    I also did reinsert my SD card and also restarted my printer and neither fixed the problem that is in the new version 2 of Bambu Studio.
    I’m including photos of the new version, the old version and the ORCA version. so you can see that it is not my printer or card.

Yours and God Bless
------Kenny Gregar------

From: BoBoYu-1 
Sent: Thursday, March 27, 2025 9:23 PM
To: bambulab/BambuStudio 
Cc: KennyGreg ; Author 
Subject: Re: [bambulab/BambuStudio] Version 2.0.0.95 not seeing micro SD card (Issue #6235)

Hello, could you please try to remove the SD card from the printer and then reinsert it? Or try using a different SD card? It might be that the SD card is damaged, or the printer failed to recognize it

—
Hello, could you please try to remove the SD card from the printer and then reinsert it? Or try using a different SD card? It might be that the SD card is damaged, or the printer failed to recognize it.

KennyGreg, 2025-03-29:
[image]
[image]
[image]

kastamauro, 2025-04-06:
I am getting this same issue using old and new micro sd cards, using different memory size, formated as FAT32. I can see the files using bambu handy app but no in Bambu studio.

KennyGreg, 2025-04-09:
In an update.  I turned off my printerr overnight and now Bambu Studio version 2.0.1.50 sees my "storage" as well as now I get a green light on the icon.

[image]

KennyGreg, 2025-04-09:
[image]

KennyGreg, 2025-04-09:
I hope this helps for anyone else having this problem?

KennyGreg, 2025-04-11:
I've reopened this topic, cause I have no idea what is going on with "Bambu Studio version 2.0.1.50".  When I turned it off for overnight and then turned it on in the morning, Studio saw my SD card and it worked normally, I was able to see my models as well as my timelapse videos ( I thought it was fixed).  Then with leaving the printer on and closing Bambu Studio and calling it a night, When I Started Bambu Studio the next morning, again my SD card was not working on version 2.0.1.50. I turned the printer off again for about 20 minutes and when I powered it back on, Bambu Studio saw the SD card and again my models and timelapse was back.  As it stands now April 11th 2AM it still sees my SD card (storage).  I'll see what happens as the days go by.  (This doesn't happen on previous versions of Bambu Studio or on ORCA).

KennyGreg, 2025-04-11:
My version 2.0.1.50 still seems to be seeing my storage (SD card) today April 11th.  I'm just wondering if turning the printer off for at least 30 minutes and then turning it back on might have something to do with it.  Possibly because of capacitive charge and discharge it takes that long for the memory of the printer to actually clear and reset? I suggest to try that and see if that is the case.  Please post if it is.

KennyGreg, 2025-04-11:
April 11th update.  With Bambu_Studio_win-v02.00.01.50 a quick on-off of the P1S printer doesn't restore the storage (SD card).  But a shutdown of the printer for 5+ minutes does restore the P1S seeing the storage  SD card.  That's not saying it works correctly.  If you try to download an early timeline video, you'll lose contact with the storage (SD card0 again. So it's definitely a bug with the Version 2.0.0.95 and the version 2.00.01.50 of Bambu Studio.

KennyGreg, 2025-04-11:
[image]

walterwongbbl, 2025-04-14:
Hi. @KennyGreg We will fix this issue in the next version.

walterwongbbl, 2025-04-14:
Hi.  @KennyGreg   
I apologize for the confusion. Do you mean that you will temporarily be unable to see the files stored on the SD card when quickly switching the P1S printer?

walterwongbbl, 2025-04-14:
@KennyGreg To help resolve this issue, could you please provide your printer logs?

walterwongbbl, 2025-04-14:
Hi. @Phil-Barker 
Thank you very much, we are currently monitoring this issue.

Phil-Barker, 2025-04-14:
[debug_Mon_Apr_14_07_54_59_40640.log.zip](https://github.com/user-attachments/files/19730984/debug_Mon_Apr_14_07_54_59_40640.log.zip)

This is the log file from opening bambu studio today
Not sure if it's what you need - if you need a log from the printer itself I'll need instructions on how to get that

Issue still persists

[image]

[image]

Same printer via mobile app (iOS)

[image]

I decided this morning that it's becoming so frustrating not being able to just repeat a previous print from bambu studio that I would take the hit to reformat the SD card - after deleting current partition, and creating a new one, formatting as FAT32 - then restarting the printer it's working in bambu studio it's no longer showing abnormal storage - so I thought I might have fixed it

However it now just hangs on this

[image] and never actually shows the list.
I've sliced a print and sent it to the printer which all went fine, but the storage screen still just hangs on `loading file list...`

and even better now it won't work either through the mobile app

[image]

what a complete mess!

Phil-Barker, 2025-04-14:
Additionally - after going through the process of reformatting the SD card, sending one sliced print to it - after I reboot the printer the abnormal storage message returned in bambu studio

[image]

KennyGreg, 2025-04-14:
**Walter** In addition could you please, please, please fix the return key on the numbers part of the keyboard.  I do a lot of editing with the size, location, and rotation, using my numbers part on the right hand side of my keyboard and out of habit I hit the enter key there and forget it doesn't work on version 2.0.0.95.

Phil-Barker, 2025-04-14:
> Walter - If I turn the printer off and immediately turn it back on, it still doesn't see my storage (SD card). However; if I turn it off and leave it off for at least 4+ minutes, then turn it back on, it will see my storage (SD card) and show me my models and Timelapse. But if I try to take a download of an old timelapse, it fails and the printer no longer sees the storage (SD card). I would have to turn the printer off for another 4+ minutes for it to see my storage (SD card) again. I don't know how to find my printer logs to be able to send them, could you tell me please how to do it?

I had to do some deep googling to find this out too

Printer logs are stored (ironically) on the SD card in a folder called `/log`

I have raised a direct support ticket and attached all of my studio logs, printer logs, screenshots, firmware versions etc...

KennyGreg, 2025-04-14:
**Thanks Phil.**  I had to laugh about it being on the SD card.  I'll look but I can't guarantee them anything will be there pertaining to it not working.

KennyGreg, 2025-04-14:
**Walter**Here goes what I think might be log files.  Please remember that they are coming off an SD card that sometimes works and fails when i access it with version 2.0.0.95 or 2.0.1.50.

KennyGreg, 2025-04-14:
logs

[Desktop.zip](https://github.com/user-attachments/files/19734146/Desktop.zip)

KennyGreg, 2025-04-14:
I'm getting too old for this s***...........stuff

walterwongbbl, 2025-04-15:
@Phil-Barker  Unfortunately, Studio's logs cannot provide more information and require printer logs to analyze this issue.
https://wiki.bambulab.com/en/X1/troubleshooting/how-to-upload-log

walterwongbbl, 2025-04-15:
Hi. @Phil-Barker @KennyGreg  

The engineers responsible for the SD card have provided some wiki pages for you to try out. Once we obtain the printer logs, we will begin analyzing the issue.
https://wiki.bambulab.com/en/general/micro-sd-faq

Phil-Barker, 2025-04-16:
> [@Phil-Barker](https://github.com/Phil-Barker) Unfortunately, Studio's logs cannot provide more information and require printer logs to analyze this issue. https://wiki.bambulab.com/en/X1/troubleshooting/how-to-upload-log

Thanks - I have filed a support ticket US250414314002 with full printer logs
Have had a reply saying there's nothing wrong with the SD card speed and printer logs seem fine

As @KennyGreg did I rolled back to v1.10.2.76 and everything works fine - this is 100% a bug in V2+ of bambu studio

[image]

KennyGreg, 2025-04-16:
**Phil**  Bambu Labs have put out an update to version 2.0.2.57.  They fixed the enter key mot working on the number pad, but unfortunately they still have the same problem with the SD card.  Perhaps they'll address that in the next update.  Here's hoping they don't break some other part when they fix the SD card.

KennyGreg, 2025-04-16:
I find using ORCA excellent at inserting text into my projects.  Its so much simpler then Bambu Studios.  It doesn't require the extra steps of converting into a negative part for indented text.  Orca does that automatically from the UI. The problem I'm finding with Bambu Studios is they over complicate the actions needed to perform some things.  Properly developed software is supposed to make it easier for the user to use.  But since they make such a great printer, I would like to be able to stick with them and help make Bambu Studio a good Slicer.  They're just making it difficult with trying to explain the problems they have.  It sounds like they're just kids, learning it as they go along.

SonOfAbaddon, 2025-04-22:
I can confirm that for me on my P1S I also cannot see the content of my SD card. v1.10 is OK but all 3 2.XX releases so far were not working. I tried swapping the card for another one and reformatting both in the printer. still not able to use the content of the cards. skipping back to Studio 1.10 solves the issue.

JohnSmithCYM, 2025-04-22:
Can anyone post a ticket and upload printer's log for us?We still need more information to detect this problem.
https://wiki.bambulab.com/en/X1/troubleshooting/how-to-upload-log

JohnSmithCYM, 2025-04-22:
@SonOfAbaddon It seems your problem should be related more with Studio. Can you offer your log after these steps?
- Click into configuration folder
- Close Studio and open "BambuStudio.conf"
- Change key "serverity_level" value to "trace" like pictures.
[image]
[image]
[image]
Then connect the printer and try again, send the log file to us.
Thanks for your feedback again.

SonOfAbaddon, 2025-04-22:
[log.zip](https://github.com/user-attachments/files/19851669/log.zip)

Here you are.

SonOfAbaddon, 2025-04-22:
> [@SonOfAbaddon](https://github.com/SonOfAbaddon) It seems your problem should be related more with Studio. Can you offer your log after these steps?
>
>     * Click into configuration folder
>
>     * [...]

here are the requested logs

[logs.zip](https://github.com/user-attachments/files/19851751/logs.zip)

Phil-Barker, 2025-04-23:
FYI we've been saying it's a studio issue from the beginning

- works on mobile iOS app
- works on all bambu studio versions before v2.0
- works locally on P1S/P1P display
- works in OrcaSlicer (any version including latest)
- can SFTP into the machine and read the SD card
[image]

- does NOT work in bambue studio v2.0+

I notice SonOfAbaddon uses a sandisk SD card, so do I, and so do the people on the support forum - so that might be a pointer though I doubt the make of the SD card would make a difference as the printer itself can read it fine, seems to be some sort of networking issue

I have also enabled trace logging and attached new logs if they help

[debug_network_Wed_Apr_23_11_00_08.log.zip](https://github.com/user-attachments/files/19865164/debug_network_Wed_Apr_23_11_00_08.log.zip)

Just looking through the logs I can weirdly see an error on the `get_user_tasks`
`Slic3r::NetworkAgent::get_user_tasks error: network_agent=000001FD2E258FC0, ret=0, http_body=` yet the http_body is valid JSON and does indeed include all the models on my SD card, so it looks like the attempt to get the models over the network is correct, but the task errors out - don't know if that helps

SonOfAbaddon, 2025-04-25:
@JohnSmithCYM what did you guys find in the logs handed over by Phil and myself?

I just installed BS2.0.03 and the issue is still there so this is something brought in with V2 and not fixed at all. It seems recommending Bambu blindly to friends and family is off the table for now...

If this persists I will try moving over to Orca (as long as the "security" features allow a way to do so).

SonOfAbaddon, 2025-04-25:
Addition: Using "Send" to send files to the device is also NOT WORKING! The files sent over are not shown in the mobile app. So I cannot prepare stuff to be printed later via the mobile app also. :( This is getting out of hand!

Phil-Barker, 2025-04-29:
@KennyGreg FIXED in v2.0.3.54

I wasn't hopeful as it's not mentioned in the fixes/changelog, but I installed anyway to see if I could dry with AMS 2 from bambu studio (you can't) but noticed SD card was showing as healthy, now works fine

[image]

Phil-Barker, 2025-04-29:
I should add - after looking through git commits and not seeing much that looks relevant I did also upgrade my P1S firmware to 01.08.00.00 just before upgrading bambu studio - so could also have been a firmware fix I guess

SonOfAbaddon, 2025-04-29:
> [@KennyGreg](https://github.com/KennyGreg) FIXED in v2.0.3.54
> 
> I wasn't hopeful as it's not mentioned in the fixes/changelog, but I installed anyway to see if I could dry with AMS 2 from bambu studio (you can't) but noticed SD card was showing as healthy, now works fine
> 
> [image]

Did you do some other stuff other than installing the new version? I just did a complete uninstall with removing userdata and clearing the AppData folder but no dice as before.

KennyGreg, 2025-04-29:
Thank you.  It gave me a message to either format or change my SD card.  I formatted it and all was working with the storage,  As an upgrade I put in a 256 GB sd card formatted as fat 32 and the storage has been working fine.  I can see my models as well as my timelapse.

Phil-Barker, 2025-04-29:
> 
> [@Phil-Barker](https://github.com/Phil-Barker): Did you do some other stuff other than installing the new version? I just did a complete uninstall with removing userdata and clearing the AppData folder but no dice as before.
> 
> Edit: are you using OrcaSlicer as well? on the X1 printers I read about issues using the 1.08 FW as they implemented their "security features" to lock out 3rd party stuff.

I didn’t do anything else just updated my P1S firmware and Bambu studio and it started working

I did use orca slicer while Bambu studio was playing up but don’t like it as much. It started putting supports in on places I know didn’t need them and the supports it added were impossible to remove

Glad Bambu got it working cos for over a year I’ve just reprinted the same models from SD card for over a thousand hours without issue. This was painful to work around

KennyGreg, 2025-04-30:
I have a P1s and my firmware is 01.07.00.00, and it says it's the latest version.  Version 01.08.05.00 from what I see is for the X1 series printer.  My version of Bambu Studio is 2.0.3.54 and all seems to be working (Knock on wood) so far.  [@Phil-Barker] am I missing something?

SonOfAbaddon, 2025-05-01:
@KennyGreg there was a 01.08 release for the P1S a few days ago, likely with the same limitations the community noticed on the X1 series.

Phil-Barker, 2025-05-01:
I'm on the 01.08 firmware on my P1S - just tried orca slicer and doesn't seem to be any issues, connects fine, can send a print.

@KennyGreg there's definitely a 01.08 - it's added the ability for P1* printers to dry with the AMS 2

[image]

KennyGreg, 2025-05-01:
I just got the update of the 1.08.00 firmware.  After I downloaded and waited for the OK, I put the printer through a "calibration", When I started up the Bambu Studio and prepared to print, it stated in red letters that my SD card was full and to either format it or replace it.  I chose to format it, but since I had installed a 2 GB SD into it, it was taking too long to do the format, so I turned the machine off, reinstalled the original SD card that came with the machine and formatted that.  All is working fine now.  Maybe doing a "calibration" after the firmware update is what's needed for the printer to accept the new firmware?  For mew the Calibration takes 26 minutes, but it gives me time to have a good sit down meal.

Phil-Barker, 2025-06-11:
It's back again

[image]

[image]

*sigh*

and now even orca slicer errors

[image]

and despite only being on 01.08.02.00 I get errors in both bambulab and orca slicer for the HMS screen related to the new firmware security

[image]

Really longing for some stability right now 😢

KennyGreg, 2025-06-11:
What I did was downgraded my firmware back to (01.08.01.00)
Why Bambu Studio has denied other open source slicers from working with their machines makes no sense.  I've seen other companies do this and watched them loose their customers to the point where they have to either rebrand and not do that any more, or they close down the entire department and stop making the product.

Phil-Barker, 2025-06-17:
They seem to have maybe fixed the MQTT thing now - but I'm back to storage abnormal, can't print from storage through bambu studio, seems to work ok through the mobile app

I've pretty much given up on hoping there will be any stability since the big v2 update
I might try a new SD card but I doubt it will help

It's ruined my workflow as I print repeatedly the same models for ebay/etsy sales straight from SD card, now I'm having to recreate them in bambu studio every time 

I never thought I'd move away from Bambu because of how reliable the P1S has been but all this software nonsense is making me feel like I might

KennyGreg, 2025-06-17:
What version Bambu Studio are you using? I just upgraded to version 2.1.1.52  My SD card seems to be working fine, although I haven tried printing from it.  I usually export my creations both as an "stl" file an a "3mf" file.  If I need to print anything again I call up the "stl" file.

janknobloch, 2025-10-25:
+1 (2.3.0.70)