bambulab/BambuStudio #6241: Just get a blank page with the letters 404 when clicking the "Online Models" in the left hand panel

gudmundur-hardarson, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

Bambu Lab P1S

### How to reproduce

Open Bambu Studio
Click "Online Models"

### Actual results

[image]

### Expected results

I expect to go to the online library of available model like in the app

### Project file & Debug log uploads

no project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gudmundur-hardarson, 2025-03-29:
Just upgraded to 2.0.1.50
This is how it looked on initial startup after installing the network plugin.

[image]
But I still get the 404 when í click the online models

[image]
or even if I go into the print history?

[image]

gudmundur-hardarson, 2025-03-29:
Ok, I seem to have found what was causing this. 
in the preferences there was no language set:

[image]

I never changed this but perhaps it's due to the fact that I live in Iceland and the application attempts to do some automatic language config and doesn't anticipate Icelandic :)

Things seem to work now.

realzorro, 2025-04-07:
Thank you for your reminder, we will optimize this design.