bambulab/BambuStudio #6244: Prime volume missing  from new 2.0.0 version.

whoffarted01, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

It is a trend across multiple machines.

### Printer

BBL A1

### How to reproduce

go to other settings, it is missing. I know it adjust based on the flush volumes but it doesn't work properly. it is very hard to tweak the size.

### Actual results

make it work properly or let me know what I did wrong.

### Expected results

Should be and to adjust prime tower to move it around objects and not waste filament.

### Project file & Debug log uploads

there is no need for this.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-28:
Thank you for your feedback. Could you take a screenshot or record a screen capture to describe your problem?

bambu-user2025, 2025-03-28:
[image]
I also have the same issue with last 2 updates. I have figured out how to adjust Prime Tower width but there is no option for Prime Volume anymore.
Previously, it was listed in Prime tower settings seen in the screenshot on this page: https://wiki.bambulab.com/en/software/bambu-studio/parameter/prime-tower

zhimin-zeng-bambulab, 2025-03-31:
@whoffarted01  This parameter has now been adjusted to a filament parameter. As shown below
[image]

SiJiL82, 2025-08-08:
This page needs updating to include this change: https://wiki.bambulab.com/en/software/bambu-studio/parameter/prime-tower