bambulab/BambuStudio #6245: "Print Infill First" is either misnamed or has a bug on top layer

ctadlock, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S and X1C

### How to reproduce

1. Use any model with a flat top (like a cube)
2. Enable "Print infill first"

### Actual results

The inside of the top layer is printed first. I think most people would not consider this infill. Note the bottom layer still prints walls first and then the inside.

[image]

### Expected results

Either the setting is misnamed, or the outside of the top layer should be printed first before the inside like the bottom.

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ctadlock, 2025-03-28:
The docs state `The infill will be printed first, followed by the inner and outer walls. “Infill” here refers to all horizontal lines, including sparse infill, internal solid infill, top surface, bottom surface, and bridge.`, but that isnt what is happening either. The bottom layer walls are being printed before the inside.

What the bug here is depends on the intention of this setting. 

Personally I believe it should just be real infill, not the top and bottom layers. The problem is the hardened inside pushes out the walls when there is only one wall on top and bottom creating a jagged effect.

[image]

ctadlock, 2025-03-28:
Bottom walls being printed before inside when "print infill first" is enabled.

[image]

XunZhangBambu, 2025-04-09:
@ctadlock  In first wall, we always print wall first to get better adhesive. And this behavior is not determined by the option