bambulab/BambuStudio #6246: AMS not detected in Calibration

timtheprinter, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

In Bambu Studio, when I click the Calibration tab, the AMS filaments are not detected or displayed. I tried clicking the new button beside "AMS and nozzle information are synced", but that didn't do anything.
The only filament for calibration that is displayed is the external.
The AMS is showing up in the device tab.

Here's what I see in the Calibration tab (no AMS):
[image]

Here's the device tab (Shows AMS connected):
[image]

### Printer

A1

### How to reproduce

1. open Bambu Studio
2. Click the Calibration tab
3. observe the AMS filaments missing...

### Actual results

The AMS does not appear as an option for calibration:

[image]

### Expected results

Expected to see the 4 AMS filaments listed (as they were in previous versions)

### Project file & Debug log uploads

There is no project file. This is in the core software.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Allen202012, 2025-03-28:
have the same issue here too #6218

zuliano, 2025-03-28:
The same here, if i upload filament ams seems to be synced and all filaments appear.
But when I start calibrating it's not starting calibration print and AMS sync error appears after few minutes.

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab