bambulab/BambuStudio #6247: Studio 2.0 caused printing error

Ochre-Me, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

A1

### How to reproduce

Bambu Studio would only import geometry, not the project

### Actual results

As a result the printer miss printed the 3 and higher layers

### Expected results

proper print

### Project file & Debug log uploads

[Trim 1 to13-L v11.zip](https://github.com/user-attachments/files/19496967/Trim.1.to13-L.v11.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-28:
[image]
Hello, I have checked and the slicing is normal, so can you take a screenshot or upload a GIF to explain what the problem is？

Ochre-Me, 2025-03-28:
Thank you for the response. I found the problem, the update changed the nozzle settings. When I reset them to the correct settings everything worked.

Once again thank you,

Andrew

604 905 9082
***@***.***

image.png (view on web)<https://github.com/user-attachments/assets/cea9ecc4-3885-461b-8518-c2a68beec32c>

image.png (view on web)<https://github.com/user-attachments/assets/cea9ecc4-3885-461b-8518-c2a68beec32c>