bambulab/BambuStudio #6250: When doing Calibration within Bambu Studio 2.0.0.95 I cannot See AMS Slots only External

Rogue2k0, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

When doing a Filament Calibration within Bambu Studio 2.0.0.95 I cannot See AMS Slots only External - YET, I can see the AMS Slots and pick a filament from the AMS when Preparing a Build Plate.

There is NO Project, I am not trying to print anything, I have replaced 2 spools of filament and cannot Calibrate them. It worked 1 hour prior.

[image]

### Printer

Bambu Lab X1C

### How to reproduce

Go to Flow Rate Calibration in Bambu Studio

### Actual results

NO AMS Spools appear

### Expected results

Expect my AMS Slots to be visible and selectable

### Project file & Debug log uploads

NO Project File as I am NOT printing anything

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Rogue2k0, 2025-03-28:
NOTE: The issue does not exist in ORCA Slicer

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab