bambulab/BambuStudio #6253: Since Update to Bambu Studio

wmprintsgb, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

Since update, all pre existing programs."3mf files", saved with support "Normal (default)" now open with Tree (Default) support. I loaded the same file and took a snap shot in both versions. Sample file "Insert" added

[image]
[image]
[image]

### Actual results

Support type settings change

### Expected results

Setting to match the original file settings

### Project file & Debug log uploads

[INSERT.zip](https://github.com/user-attachments/files/19503970/INSERT.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

wmprintsgb, 2025-03-28:
Image order and description, 
1, Previous version of INSERT.3mf
2. New Version open new file
3. New Version Open INSET.3mf

Version comparison:
Bambu Studio v1.10.2.76
vs
Bambu Studio v2.0.0.95

NothFie, 2025-03-31:
@wmprintsgb Thanks for your report. We apologize for the inconvenience caused. This is a normal occurrence, as we have changed the default Support Style from NORMAL to TREE support in the new version. If you would like to use normal support, you will need to adjust the style to NORMAL in the Studio and save your changes again.