bambulab/BambuStudio #6254: Unable to calibrate flow due to the AMS slots are not available on Calibration page

M-A-B007, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Additional system information

Device name	MichalovoPC
Processor	13th Gen Intel(R) Core(TM) i9-13900HX   2.20 GHz
Installed RAM	32.0 GB (31.7 GB usable)
Device ID	D09135D2-9478-4574-A1F1-3C2E9ABF6B1A
Product ID	00342-21095-20925-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
-----
Edition	Windows 11 Home
Version	24H2
Installed on	‎1/‎4/‎2025
OS build	26100.3476
Serial number	PF4DZNGT
Experience	Windows Feature Experience Pack 1000.26100.54.0
-----

### Printer

X1C

### How to reproduce

1. have X1C printer with AMS connected. 
2. update BambuStudio to latest
3. perform Dynamic FLow calibration 
     - this time I was able to see all the AMS slots. New graphics - ok. Was able to select 2 AMS slots and launch calibration for them. Printer printed the calibration pattern.
     - here the final screen with results was skipped. Well, might happen I overlooked it. so I run the calibration again.
4.  Try to run the dynamic flow calibration again.
----
does not relate to a particular model or project. It occurs even when no project is open.
----

### Actual results

- The calibration dialogues do not show the AMS slots.
- The calibration dialogues only show the direct slot.
- As a result, I am unable to calibrate flow and dynamic flow for materials in AMS.   No bueno. All printing is effectively blocked.

### Expected results

- on Calibration page see all the AMS slots with material spools inserted.
- To have possibility to refresh the info from AMS.
- be able to calibrate

### Project file & Debug log uploads

[log-screenshots.zip](https://github.com/user-attachments/files/19504155/log-screenshots.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

ondrovic, 2025-03-28:
same result flow dynamics and flow rate

[image]

[image]

[image]

[image]

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab