bambulab/BambuStudio #6255: Plate Selection click on Picture

currydani, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Move mouse over new Build Plate selection picture
2. Click on the picture (List opens)
3. Move mouse to the list
4. List disapears while moving mouse from picture over the plate name

### Actual results

You can't select the build plate by clicking the picture. The list opens but you cant reach it because it disapears while moving the mouse towards it. You can only select the plate by clicking the name of the plate itself.

### Expected results

Keep the list of plates open as long as the mouse is on the Build plate selection field (picture as well as the plate name, everything inside the grey boarder of that field)

### Project file & Debug log uploads

[Plate Error.zip](https://github.com/user-attachments/files/19504905/Plate.Error.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-03-31:
Hello, your idea is reasonable, but currently it is not technically feasible to directly preview the specific disk image when sliding the drop-down menu options. I'm very sorry~

currydani, 2025-03-31:
I think i didn't explain it properly. Here i got a video that shows what i mean. I guess it should be expected, that if you click onto the picture of the plate to open the drop down, you should also be able to move the cursor to it without it disappearing. You understand what i mean?

https://github.com/user-attachments/assets/a2f6d6ba-2fd3-46a8-ba82-831a8d001c7a

Haidiye00, 2025-03-31:
Hello, if the enlarged image is displayed and the dropdown menu still doesn't disappear, some people may think that all the images for the dropdown options are the same, which may also cause confusion.

currydani, 2025-03-31:
That's not what I meant. It's fine for the picture to disappear, but it is not for the drop-down list to do that without an action.
Can you reproduce my problem?

1. Move mouse over picture of plate
2. Click the plate Picture (picture disappears, drop down opens)
3. Try to select a plate in the list

Can you verify that the same thing happenes to you?

Haidiye00, 2025-03-31:
Hello, I roughly understand what you mean. Clicking on a dropdown list through an image should be equivalent to clicking on the dropdown list, and it should not disappear without clicking again. Let me analyze the reason~

Infraviored, 2025-03-31:
Issue is a disappearing drop-down. 
The drop-down is unusable as by moving from dropdown over to the actual drop-down items, the selector disappears. 
No option to select an entry.

Haidiye00, 2025-04-01:
[image]
Hello, this is the result of my repair. Thank you for your suggestion~

closed 2025-04-16 by Haidiye00