bambulab/BambuStudio #6257: Bambustudio 2.0 is shit

pollosud, 2025-03-28:
### Bambu Studio Version

2.0

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

The new version of Bambustudio is unusable.It is full of bugs.

### Printer

A1

### How to reproduce

I t is  shit

### Actual results

A new version

### Expected results

Shit

### Project file & Debug log uploads

shit

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-28:
Sorry, may I ask what problem you are facing? Please provide a screenshot and a description.

pollosud, 2025-03-28:
It crash every 2 times  and it slowd down all my pc.Noe I use orcaslicere.It is best

Haidiye00, 2025-03-28:
Hello, please use the evevything software to search for the bambuStudio folder and provide the dmp and logs of the log folder crash. Thank you. We will analyze and solve it~

pollosud, 2025-03-28:
Now i use Orcaslicer, Bambustudio is fucking

Haidiye00, 2025-03-29:
Hello, may I ask if your computer's chip is ARM？ It may be due to this reason

closed 2025-06-03 by Haidiye00