bambulab/BambuStudio #6260: I can't send print job

asher8554, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Window 11 24H2

### Additional system information

12th Gen Intel(R) Core(TM) i7-12700   2.10 GHz
RAM	32.0GB(31.7GB 사용 가능)
시스템 종류	64비트 운영 체제, x64 기반 프로세서
Windows 11 Pro
24H2
‎2025-‎01-‎26
OS 빌드	26100.3624
경험	Windows 기능 경험 팩 1000.26100.66.0

### Printer

Bambu Lab A1 Mini

### How to reproduce

[image]

### Actual results

Doesn't work

### Expected results

Send print job

### Project file & Debug log uploads

https://github.com/user-attachments/assets/25c0386b-27e2-47e5-b747-1df53976c61f

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-29:
Hello, please try using the latest version
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.5

asher8554, 2025-03-29:
Please check link
[image: image.png]

Haidiye00, 2025-03-29:
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
Sorry, i missed copying a 0

BoBoYu-1, 2025-03-31:
Thank you for your feedback. Please set the nozzle type and diameter in the "Printer Parts" section under the "Device" page. Try initiating the print again.
[image]

asher8554, 2025-04-01:
Thanks for reply. However A1 mini does not work.
Recently I use old version.

please check  below image.
[image: image.png]
[image: image.png]

XinZhangBBL, 2025-04-10:
@asher8554 Sorry for the issue. Could you please reboot your printer, and retry sending print

XinZhangBBL, 2025-04-25:
@asher8554 Halo,  The problem solved or not ? If there is no reply, we will close this issue。
If not solved, please try new version.
https://github.com/bambulab/BambuStudio/releases

[image]