bambulab/BambuStudio #6262: AMS slot-selection not working - AMS useless under macOS since Firmware update

Surfable, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

15.3.2 (24D81)

### Additional system information

Apple M4

### Printer

Bambu Lab X1C

### How to reproduce

Slice anything. Try to send Print. Try to select Filament Slot from AMS.

### Actual results

The selection menu before send wont let me choose any slot. Selection is not clickable however seems accessable but wont accept any click - remains on slot 1. Desired filament needs to be loaded to slot 1 manually in order to print at all.

### Expected results

Lets me switch between slots in case its the same material.

### Project file & Debug log uploads

[Print.3mf.zip](https://github.com/user-attachments/files/19506551/Print.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-28:
Thank you for your feedback. This issue has been fixed. Please update to the latest version V2.0.1.50 and try again.
https://bambulab.cn/zh-cn/download/studio

Surfable, 2025-03-28:
Thanks bud. Working!

closed 2025-03-28 by Haidiye00