bambulab/BambuStudio #6264: Updates, installation

Brocodd24, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Additional system information

The new updates now require a separate update download. Updates cannot be easily installed on a corporate device without permission to install new programs. I now need to submit an update request to my IT department for minor updates. This was never the case before updating to 2.0. 

### Printer

A1

### How to reproduce

1. Downlod Update
2. Open download
3. Install

### Actual results

Error opening file for writing

### Expected results

update is installed.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-31:
Hello, you can try upgrading to the latest Studio version V2.0.1.50 available on the official website and see if it works better:
https://bambulab.cn/zh-cn/download/studio

lanewei120, 2025-04-16:
Hello @Brocodd24 

does this issue still there?
could you give a snapshot of your errors

thanks

Brocodd24, 2025-04-21:
Yes, it's still an issue.  The problem is my work computer is controlled by our IT department and requires approval for new software installations.  It's never been an issue It's never been an issue to perform updates with Studio V1.  Since the updates are are requiring a seperate download, my computer sees it as new software. 

[image]

[image]

Brocodd24, 2025-04-21:
> Hello, you can try upgrading to the latest Studio version V2.0.1.50 available on the official website and see if it works better: https://bambulab.cn/zh-cn/download/studio

This wont help, downloading new software requires administrative privileges.  I've never needed Admin privilages for software updates with V1.  My IT dept. will shoot me if I call them to perform every so

closed 2025-04-21 by Brocodd24