bambulab/BambuStudio #6268: BambuStudio -> Device -> Status: synchronising rating information has failed

rmilkowski, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4

### Printer

X1C

### How to reproduce

Print something from BambuStudio and switch to Device->Status tab to observe the progress (I had video enabled too). Once the print is done in the bottom part (Printing Progress window) at the end I get in red: You have completed printing the mall model, but synchronising rating information has failed.

I'm not next to the printer right now but I believe you get a pop up message on the lcd asking to rate the print. The message never shows up for me in BambuStudio though.

### Actual results

Error message printed in red after each print completes in Bambu Studio. This is unnecessary.

### Expected results

 I think studio should either ask for the rating or not print the message in red as an error (or not print it at all).

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rmilkowski, 2025-03-28:
I've just noticed that in the bottom right corner there is a green Retry button, I clicked it and then I do get "How do you like this print file?" along with stars and the error message is gone. However, unless I click on the Retry button I always get the red error first (or at least most of the time - don't remember).

DanBao-Bambu, 2025-04-10:
If such a thing happens again, can you send your log?
Please refer to: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

rmilkowski, 2025-05-02:
I did some prints with the latest BambuLab 2.0.3.54 and the issue did not reoccured.

closed 2025-05-02 by rmilkowski