bambulab/BambuStudio #6270: Filament Advanced Toggle button doesn't do anything

ondrovic, 2025-03-28:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

n/a

### Printer

BambuLab printers

### How to reproduce

1. Under project filaments
2. Next to any filament click the ...
3. Select edit
4. On the top right hand side try and toggle the advanced toggle button

### Actual results

Nothing happens when toggling the button

[image]

### Expected results

The button should enable some advanced features or be removed to reduce confusion in the ui 

### Project file & Debug log uploads

project: n/a
logs: n/a

### Checklist of files to include

- [x] Log file
- [x] Project file

JeffGoldner, 2025-03-29:
Do you have Developer Mode turned on?  (Settings / Preferences depending on OS)

ondrovic, 2025-03-29:
> Do you have Developer Mode turned on? (Settings / Preferences depending on OS)

I do have developer mode turned on, even if it's turned on and that's the cause the button should be made invisible so it's not there, that way it's not being show on the page as it clearly does nothing

````
// something like this 
if (devMode) {
 advancedToggle.visible == false
}
```