bambulab/BambuStudio #6273: When running Calibration, all filament slots in AMS showing Incompatible

FlewByYou, 2025-03-28:
### Bambu Studio Version

2.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.2

### Printer

Bambu Lab X1C

### How to reproduce

1. Make sure AMS is sync'ed
2. Go to **Device** tab and make sure all filament slots in AMS show up
3. Click **Calibration** tab and choose either **Flow Dynamics** or **Flow Rate** calibration
4. All filament slots are showing up as **_Incompatible_** in both **Auto Calibration** or **Manual Calibration**

### Actual results

All filament slots in AMS showing up as **Incompatible**

[image]

**

### Expected results

Filament slots should show up with their respected filament brand/type

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19508683/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-03-29:
Hello, please try using the latest version
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.5

FlewByYou, 2025-03-29:
> Hello, please try using the latest version https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.5

I am using the latest version.

jimenezff, 2025-03-29:
I'm still having this issue with v2.0.1.50

a-dam-berger, 2025-04-03:
I have this same issue. Please fix this.

SterMouth, 2025-04-04:
Issue still occurs with V2.0.1.50