bambulab/BambuStudio #6274: Unable to edit existing flow calibration result

gary-wlg, 2025-03-28:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.1

### Additional system information

When trying to edit an existing Flow calibration value, you receive an error "Within the same extruder, the name 'calibration_name' must be unique when the filament type, nozzle diameter and nozzle flow are identical. Please choose a different name.

The only way to modify an existing value now is to delete the old calibration entry and create a new one. 

### Printer

n/a

### How to reproduce

Under the calibration tab, click "Manage Results", then click edit beside an existing calibration entry.  Change the value, and click save.  You will get the error message.

### Actual results

[image]

### Expected results

Expected that the existing calibration entry is updated with the new K factor.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

FlewByYou, 2025-03-28:
Same issue on macOS 15.3.2

zhimin-zeng-bambulab, 2025-03-31:
Thanks for reporting this issue, we will fix it in the next version. You can delete the the value first and then create it again.