bambulab/BambuStudio #6275: Cannot type in a 'D' letter in the text tool

ageitgey, 2025-03-28:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1

### Printer

Bambu Lab X1C

### How to reproduce

1. Load a part in Bambu Studio
2. Use the text tool to input a word like "Dog" with a capital "D" in the "Input text" box

### Actual results

Other letters can be typed, but "D" can not be typed.

[image]

### Expected results

You should be able to type "D"

### Project file & Debug log uploads

This happens with any project file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

darth-pika-hu, 2025-03-28:
LOL. Same here. A workaround solution is using CMD+C and CMD+V to copy and paste the letter D.

Haidiye00, 2025-03-29:
Hello, this is a bug. We will fix it in the next version.
At present, you can input it elsewhere and then copy it in. We apologize for any inconvenience caused

Ddcr13, 2025-04-01:
Same problem here

I have to use Caps Lock instead of Shift.
The command "Shift+D" is a shortcut to "Delete All". 

[image]

Haidiye00, 2025-04-01:
Hello, the basic reason is caused by shift+d. We will change it to ctrl+shift+d. Actually, this shortcut key is rarely used to delete all of it. I usually use Ctrl+A to select all and then delete them

closed 2025-04-16 by Haidiye00