bambulab/BambuStudio #6280: Failed to install network plugin, MacOS 15.4

JeffGoldner, 2025-03-28:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4 RC

### Additional system information

Mac Studio M2 Max, 32GB

### Printer

P1S

### How to reproduce

Install 2.0.1.50
Launch it
You get a popup asking to install the network plugin
Fails after first launch of Bambu Studio
Help link is not helpful for MacOS users as it points to a Windows download

I have 1.0 still saved on my system just in case

### Actual results

Closed Bambu Studio 2.0.1.50
Reopened BS, works fine now, so first launch issue

### Expected results

No errors

### Project file & Debug log uploads

Not using any project file yet

### Checklist of files to include

- [ ] Log file
- [ ] Project file

FlewByYou, 2025-03-28:
I ran into the same issue at first launch. I then removed /Users/<user>/Library/Application Support/BambuStudio/plugins folder and it re-downloaded the plugin w/out any issues thereafter.

darth-pika-hu, 2025-03-28:
Just reboot the Mac.

JeffGoldner, 2025-03-29:
As I stated, it worked fine after restarting Bambu Studio. No need to delete anything or reboot the system.

FyrbyAdditive, 2025-03-29:
I had the same here today, additional information in the form of the error message attached. It does work after restarting Bambu Studio, however people will not realise this or know to look here for this info probably.

[image]

darth-pika-hu, 2025-03-29:
> I had the same here today, additional information in the form of the error message attached. It does work after restarting Bambu Studio, however people will not realise this or know to look here for this info probably.
> 
> [image]

Restart Mac

Qjeezy, 2025-03-30:
Same happened to me on my Mac mini. Closed studio and reopened and all was good. 

I did notice studio is using over 16gb of memory now. The old version only used between 7-10

jpsulisz, 2025-03-30:
Tried a few solutions: restarted the computer, application, and including removing the Application Support directory. Don't have any of the products listed on the help link. This is a fresh installation on a M2 MBA and each time I login to BS I am hit with the screenshot above https://github.com/bambulab/BambuStudio/issues/6280#issuecomment-2764055251

JeffGoldner, 2025-03-30:
When I installed the 2.0 release, I kept the 1.0 version around. If you open the 1.0 version, it will install the network module OK and just work. Now, **_close_** 1.0 and open 2.0 and you get the error when it tried to install the network module. Close 2.0 and reopen and it works (for me).

Same thing happens any time I run 1.0 and then attempt 2.0 (I am NOT trying to run both at the same time). Just close it and reopen.

Haidiye00, 2025-04-04:
Okay, we will fix it~

closed 2025-04-16 by Haidiye00