bambulab/BambuStudio #6281: bambu studio 2 misses some improvements and even worsened some UI items

gitterman, 2025-03-28:
With the recent update to bambu studio some things have been improved, but why the heck do good user interface designs need to be changed without need?
Example is the build plate field which is now crippled and doesn't allow showing the full text while a mostly useless picture has been added.
Other annoyances are still there, though.
So it is still a total nag, that the print speed cannot be set before the actual print has been started. 
Why can't it be set right after an upload?
Another nuisance is the printer lamp, which should have a setting to go on when the printer is switched on and go off after a print job has finished.
It should then go on again if I press any button on the control panel.
The rationale for this is, that at power on, the first thing to do is to inspect the printer.
As print jobs always take some time, it is a waste of energy to keep the lamp on all the time after a job has finished.
It would instead be totally sufficient if the lamp goes on again for some time, when a button gets pressed, so that the printer can be inspected again properly after taking out the build plate.

Haidiye00, 2025-03-29:
Hello, thank you for your suggestion. We will make improvements. Could you please take a screenshot for some explanation? Thank you~

gitterman, 2025-03-29:
**before** (screenshot taken from your wiki)
[image]

**after update**
[image]

**Another step back is the "send print job" applet, where I cannot change the filament anymore like before.**

[image]

Haidiye00, 2025-03-29:
Hello, the issue of the consumable thread not being able to switch has been fixed. Please download https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50

gitterman, 2025-03-29:
I am on release v02.00.01.50, and it doesn't work.

[image]

Haidiye00, 2025-03-29:
Hello, there doesn't seem to be any problem with the screenshot just now. Only consumables of the same type can be selected. The reasons for not being able to choose consumables are also given tips

gitterman, 2025-03-29:
Before rev. 2 there was no problem selecting the filament on the printer page.
Right now, it only allows selecting another AMS slot in case that the selected filament runs out.
This is something completely different.
Now I MUST select the filament on the prepare page and cannot change it anymore on the "Send to Printer" applet.

Haidiye00, 2025-03-29:
Hello, can you post the effect of version 1.10. I remember ASA cannot match PLA. If you want to match, you have to modify the A3 slot to not have ASA type.

gitterman, 2025-03-29:
It is not about matching, it is about selecting the target filament for the print job on the "send to printer" applet instead of the "prepare" tab.
Unfortunately, I do not have v1 anymore.
I suppose the developers should have the old version installed, so that they can check for the differences.
Before v2 it looked like described in the bambulab wiki for bambu handy.
https://wiki.bambulab.com/en/makerworld/tutorials/printing-tutorial-on-bambu-handy.
Have a look at Step 3 "print the 1st plate"

[image]

Haidiye00, 2025-03-29:
Hello, I didn't understand the question you're asking. How did it transfer to Bambu Handy now? Can you explain again which areas are problematic

Haidiye00, 2025-03-30:
Hello, so where do you think we still need to improve. Could you please provide screenshots and relevant explanations? Thank you

gitterman, 2025-03-30:
The biggest problem to me is the .3mf overwrite.
Loading a .3mf file from Makerworld replaces the AMS filaments with those from the maker, which most likely do not match to my current setup.
This is ok for the .3mf job as it should include the setting of the make, but Bambu Studio does remember those settings for subsequent print jobs, too.
I regard this a bug.
It is like opening a Word file from someone else and then getting your font settings replaced.
Bambu Studio even replaces the filaments if I only look at .3mf file and not even print it.

Another thing which nags, is that I cannot set the default AMS slot for new print objects.
It always defaults to slot 1.
It would be nice if there was a setting to remember the last AMS slot for new jobs.

What also nags, is the printer speed setting, which says it only takes effect during printing.
For me, printing start, after the print job has been transferred to the printer and the printer starts heatbed preheating and bed levelling. 
If I change the printer speed at this point, my change is discarded, though, and I have to wait until the printer has processed its preparation tasks and really start printing, which can take some annoyingly long time.
This may be firmware related, but still Bambu Studio could remember the speed setting and send it to the printer again, when the printer status shows "printing progress".  

[image]