bambulab/BambuStudio #6282: Flushing volumes window is empty

togeis, 2025-03-28:
### Bambu Studio Version

2.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.3.2

### Additional system information

Macbook Pro 15-inch, 2018

### Printer

X1C

### How to reproduce

1. Start BambuStudio
2. klick on "Flushing volumes"
3. an empty windows opens and shows only white

[image]

### Actual results

Empty windows with white background shown instead of flushing configuration content

### Expected results

configuration options for flushing

### Project file & Debug log uploads

[empty_flushing_window.3mf.zip](https://github.com/user-attachments/files/19511810/empty_flushing_window.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

FlewByYou, 2025-03-28:
Works fine on MacBook Pro 16" M1, running macOS 15.3.2

[image]

XunZhangBambu, 2025-04-02:
@togeis  It's might be related to specific processor and system versions. Can't reproduce in MacOS 13(intel cpu) and MacOS 15(M1). Could you please provide the log files?

KilonovaRemnant, 2025-04-08:
I am having the same problem after updating Bambu Studio last night.

Software Updated to Version: BambuStudio 2.0.1.50 

Computer: MacBook Pro 14" M2
OS: MacOS 15.3.1

[image]

KilonovaRemnant, 2025-04-08:
HOW I SOLVED IT: I downloaded the Bambu Studio and put it in the Applications folder, not a sub-folder in the Applications folder.  As soon as Bambu Studio was in the Applications folder, the flushing volumes window came back.  

(✅) I tested it multiple times by opening the same file in the Bambu Studio in the Applications folder (✅).  

(❌) Then opening the same file in the Bambu Studio in a sub-folder in the Applications folder (❌).

(✅) Back to opening the same file with B Studio directly in the Applications folder (✅).

XunZhangBambu, 2025-04-09:
@togeis  In version 2.0, the flush page is implemented via wxWebView, which relies on HTML files stored in the resource path. If the application is renamed or not placed in the Applications folder, the corresponding HTML files cannot be loaded, resulting in a blank page.

jwdaou, 2025-04-12:
Nothing is working for me. I tried updating, redownloading, making sure it’s not renamed, and making sure it’s not in a subfolder. I even tried moving the “unix executable file” (located in the “show package contents”), also known as the grey icon with the “exec” in green for applications, into the applications page in finder. Still nothing unfortunately.

coolplantain9510, 2025-04-12:
Nothing was working for me. If you are reading this, I'll explain what finally worked for me. First, I moved BambuStudio to the trash, redownloaded it, and made sure it was updated to the latest version. Next, I made sure that in the applications section of finder, that it wasn't in any subfolder. Then, I made sure the application (the green icon in the applications folder with the BambuLab logo) was not renamed and was titled "BambuStudio" exactly. Still nothing was working, so this is what did it next. I left clicked on the BambuStudio icon in the applications section and clicked "show package contents". I went to the "MacOS" folder and pulled the "exec" file out and directly into the applications section. At this point, both BambuStudio and the "exec" file were in the applications section. I quit BambuStudio and opened it up again, checked flushing volumes and still nothing. I moved the exec file back into the "MacOS" folder in "show package contents" for the BambuStudio application, quit BambuStudio, opened it up again, and it finally showed up. Hope this helps.

XunZhangBambu, 2025-06-03:
@togeis @coolplantain9510 @KilonovaRemnant  Hi everyone, this has been fixed in our latest version 02.01.00.59