bambulab/BambuStudio #6283: BambuStudio does not work on ARM

Rickyrz, 2025-03-28:
### Bambu Studio Version

2.0

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 arm 

### Additional system information

I don't know if anyone has already reported it, but all Surface Pro 11 with ARM processors do not work with BambuStudios since version 1.10.1.50, this is the last version to work.
After this, nothing works, the software installs normally and opens normally, but when it asks to update the network plugin it goes up to 75% and the software turns off and no longer opens.
There is a lot of chaos on reddit about this so I don't know if it has been reported yet.
But it's frustrating that several versions have been released since 1.10.1.50 and this still hasn't been fixed.

### Printer

p1s

### How to reproduce

I don't know if anyone has already reported it, but all Surface Pro 11 with ARM processors do not work with BambuStudios since version 1.10.1.50, this is the last version to work.
After this, nothing works, the software installs normally and opens normally, but when it asks to update the network plugin it goes up to 75% and the software turns off and no longer opens.
There is a lot of chaos on reddit about this so I don't know if it has been reported yet.
But it's frustrating that several versions have been released since 1.10.1.50 and this still hasn't been fixed.

### Actual results

I don't know if anyone has already reported it, but all Surface Pro 11 with ARM processors do not work with BambuStudios since version 1.10.1.50, this is the last version to work.
After this, nothing works, the software installs normally and opens normally, but when it asks to update the network plugin it goes up to 75% and the software turns off and no longer opens.
There is a lot of chaos on reddit about this so I don't know if it has been reported yet.
But it's frustrating that several versions have been released since 1.10.1.50 and this still hasn't been fixed.

### Expected results

I don't know if anyone has already reported it, but all Surface Pro 11 with ARM processors do not work with BambuStudios since version 1.10.1.50, this is the last version to work.
After this, nothing works, the software installs normally and opens normally, but when it asks to update the network plugin it goes up to 75% and the software turns off and no longer opens.
There is a lot of chaos on reddit about this so I don't know if it has been reported yet.
But it's frustrating that several versions have been released since 1.10.1.50 and this still hasn't been fixed.

### Project file & Debug log uploads

I don't know if anyone has already reported it, but all Surface Pro 11 with ARM processors do not work with BambuStudios since version 1.10.1.50, this is the last version to work.
After this, nothing works, the software installs normally and opens normally, but when it asks to update the network plugin it goes up to 75% and the software turns off and no longer opens.
There is a lot of chaos on reddit about this so I don't know if it has been reported yet.
But it's frustrating that several versions have been released since 1.10.1.50 and this still hasn't been fixed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jpoisso, 2025-03-28:
Samsung Book with Snapdragon Elite X - ARM64  same issues.

OrcaSlicer works just fine as usual, but the last 2 Bambu releases break on the network plugin installation (force closes upon completion and immediately shuts down every time after)

Please Bambu, this issue has been known for multiple releases already.

swhook52, 2025-03-28:
I'm experiencing the same thing today. ARM64 processor. Required update to Bambu Studio resulted in a failure. Rolling back to version 1 results in failures on the network plugin install too. Bambu Studio is now unusable on ARM64.

konzolec, 2025-03-29:
Confirming the last version working is 1.10.1.50. Not sure, what you changed but please include ARM.

lanewei120, 2025-03-29:
yes,  we are working on this
hope we will solve all the issues soon

currently if you want to use 2.0, 
please skip the network plugin, and work under offline mode

Goodwine, 2025-04-02:
duplicate of https://github.com/bambulab/BambuStudio/issues/6014

lanewei120, 2025-04-07:
please check https://github.com/bambulab/BambuStudio/issues/6014#issuecomment-2782345560
for the current solution

we will fix it in next version

lanewei120, 2025-04-16:
fixed by v2.0.2.57(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57)

closed 2025-04-16 by lanewei120

side-side-project, 2025-05-18:
I'm currently on the latest version and the crash to desktop is back, but not at the network plugin stage.

It now impacts bambu and orcaslicer the same way - app loads and before it displays user data, it completely crashes (with or without a project opened)