bambulab/BambuStudio #6287: A1 Mini Missing Plate Types

AzmariaDei, 2025-03-29:
### Bambu Studio Version

v02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Multiple Computers

### Printer

A1 Mini

### How to reproduce

Go to 'Prepare' page.
Under Printer, click on the plate selection.
Only 3 options are shown in the dropdown menu. 

### Actual results

Only 3 options are shown in the dropdown menu. 'Cool Plate' and 'Engineering Plate' are missing.

### Expected results

There should be 5 results like other printers, including 'Cool Plate' and 'Engineering Plate'.

### Project file & Debug log uploads

[debug_Fri_Mar_28_20_18_08_17920.log](https://github.com/user-attachments/files/19514624/debug_Fri_Mar_28_20_18_08_17920.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-03-29:
Hello, this is not a bug. After our internal confirmation, a1mini currently does not support these two plate types. Thank you for your discovery~

AzmariaDei, 2025-03-29:
So what do I do with my cool plates that I've been using for printing miniatures?

Haidiye00, 2025-03-30:
[image]
Hello, choose this and give it a try~

AzmariaDei, 2025-03-31:
> image.png (view on web)
> Hello, choose this and give it a try~

It seems to be close enough - I just need to dial in the settings for
supports again.

Thanks! ^_^
Azmaria

Haidiye00, 2025-03-31:
Hello, if the problem is solved, can you help close the order? Thank you~

closed 2025-03-31 by AzmariaDei