bambulab/BambuStudio #6289: In Multiple plates can't choose diferent colors

miguelamartinez75, 2025-03-29:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.2

### Additional system information

CPU M1 Pro with 32 GB of memory

### Printer

Bambulab X1C

### How to reproduce

Open multi plate project, choose a color from the AMS and slice. 

### Actual results

I choose filament 1 but use 2 and 3 (choose black and use white PET-G and PLA)

### Expected results

Use the filament I had chosen 

### Project file & Debug log uploads

R.E.P.O._Robot_wip_v6

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-29:
Hello, could you please re upload the 3mf and record an operation video GIF at the same time for us to analyze the issue. Thank you

miguelamartinez75, 2025-03-30:
I sent you a reply by email, but I couldn't attach the video. Every time I tried, it told me the email was too large and couldn't send it. I sent you the 3mf anyway.

Haidiye00, 2025-03-31:
Hello, my email is zhou.xu@bambulab.com

Haidiye00, 2025-03-31:
Hello, I have received your video. Can you please send me this 3mf again? Thank you

miguelamartinez75, 2025-03-31:
Yes!

Miguel Angel

Haidiye00, 2025-04-01:
[image]
Hello, we have investigated the cause and found that you inserted two colors in these two positions. If you delete them, it will be normal~

Haidiye00, 2025-04-02:
Hello, if there are no issues, could you please help close this problem? Thank you

miguelamartinez75, 2025-04-03:
Perfecto! It works. Thank you very much

closed 2025-04-03 by miguelamartinez75