bambulab/BambuStudio #6290: STEP file Import Multibody File as Single Body

Ert850, 2025-03-29:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

HP ZBook Fury 16 G10 Mobile Workstation PC
Processor	13th Gen Intel(R) Core(TM) i9-13950HX (2.20 GHz)
Installed RAM	64.0 GB (63.6 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Select multibody file to import. Have not tested if issue is repeatable with smaller files, but seemingly happens on larger files
2. Import via any method (drag and drop, import, add+ icon, etc.)
3. Wait several minutes while slices (normal for complicated files)
4. When import, although satisfactory, all bodies are lost

### Actual results

Object imports as single body, although (in this case) it should have multiple bodies (25)

[image]

### Expected results

Imports multibody part, where individual bodies are selectable and can be edited.

### Project file & Debug log uploads

[Summer Winter Robbing Screen Left Merge Error.zip](https://github.com/user-attachments/files/19515694/Summer.Winter.Robbing.Screen.Left.Merge.Error.zip)

[debug_Wed_Mar_26_22_11_38_4828.log.zip](https://github.com/user-attachments/files/19515703/debug_Wed_Mar_26_22_11_38_4828.log.zip)

[log.zip](https://github.com/user-attachments/files/19515705/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Ert850, 2025-03-29:
File STEP which was being sliced and imported and previous GCODE showing the file is correct and is recognized in prior versions

[Summer Winter Robbing Screen PLA_REV1.gcode.zip](https://github.com/user-attachments/files/19515714/Summer.Winter.Robbing.Screen.PLA_REV1.gcode.zip)

[Summer Winter Robbing Screen Left Merge.zip](https://github.com/user-attachments/files/19515712/Summer.Winter.Robbing.Screen.Left.Merge.zip)

Ert850, 2025-03-29:
Confirmed still an issue in most recent update. Sliced same file in newest OrcaSlicer version, no error.

MackBambu, 2025-03-31:
The 3MF file doesn't contain the original STEP file, so I can't see what the issue is. Could you upload the original STEP file?

Also, you can try enabling the option "Split compound and compsolid into multiple objects" — it allows for finer splitting of the model.
[image]

MackBambu, 2025-03-31:
To traverse deeper-level solids in compound and compsolid entities, you need to enable the following option.
[image]

[image]

Ert850, 2025-03-31:
If one had clicked “Don’t show again”, is there a way to reenable that comment or access that setting? I no longer get the step file importer parameter prompting. Thank you.

MackBambu, 2025-03-31:
[image]

[image]

closed 2025-04-07 by MackBambu