bambulab/BambuStudio #6292: The ‘enter’ key on the numeric keypad

Cyano16, 2025-03-29:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1 Add primitive
2 Clic cube
3 Add dimension with numeric pad
4 clic "enter" on numeric pad
5 it doesn't work

### Actual results

The value must be confirmed by pressing the ‘enter’ key on the alphabetic keyboard.

### Expected results

The value must be confirmed by pressing the ‘enter’ key on the alphabetic keyboard.

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-29:
Hello, we have already fixed it in the internal version and expect the next version to be released. At present, you can input and click on the blank space with the mouse

guanyun-gudujian, 2025-04-16:
@Cyano16 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

Cyano16, 2025-04-16:
Super, ca fonctionne comme avant.

Merci

closed 2025-04-17 by Haidiye00