bambulab/BambuStudio #6294: Please improve the interface of the latest BambuStudio 2.0 version.请改进最新2.0版本BambuStudio的界面, to englihs

PandaBalls, 2025-03-29:
最新版软件GUI的修改，导致很多不方便的情况，并且冗余地使用了图标或者图片，导致界面人机交互变差。尤其是发送打印任务：原本的“调平”选项可以快速被勾选或者取消，现在还需要先展开高级设置才能去勾选，而且现在的switch开关远不如原本的勾选框看着直观易于辨认。

The modifications to the GUI in the latest software version have introduced several inconveniences, including excessive use of redundant icons or images, which has degraded the user interface's human-computer interaction.

A particularly problematic change is in the print job submission process:

Previously, the ​**"Leveling" option** could be quickly checked or unchecked with a simple click.
Now, users must first ​expand the advanced settings to access this option.
Additionally, the new ​switch toggle is far less intuitive and recognizable compared to the original ​checkbox design.
These changes have made the interface less efficient and user-friendly.

JohnSmithCYM, 2025-03-29:
Thanks for your feedback, we are considering some UI change in ‘Send Print’ page, to avoid mislead in some action or unconvience of some option. If we decide the modification, I will post the plan here. Thanks again.

Bluedai, 2025-03-31:
On this occasion, where is the time-lapse option now? I would like to be able to set the options without any detours every time I print. Thank you for rethinking the send print GUI.