bambulab/BambuStudio #6295: Instant Crash on Launch – Bambu Studio on Surface Pro 11 (ARM)

tobitoobitooobi, 2025-03-29:
### Bambu Studio Version

01.10.02.76 and 02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home (Version 24H2)

### Additional system information

CPU: Snapdragon(R) X 12-core X1E80100 @ 3.40 GHz   3.42 GHz
Memory: 16,0 GB RAM
Display Card: Qualcomm(R) Adreno(TM) X1-85 GPU

### Printer

Bambu Lab P1S

### How to reproduce

1. Install Bambu Studio 01.10.02.76 (or 02.00.01.50) on a Microsoft Surface Pro 11 with Snapdragon (ARM) Processor
2. Launch Bambu Studio

### Actual results

On the first launch after installation, the splash screen appears, followed by the main application window and a small dialog indicating that plugins are being downloaded. Before the progress bar completes, the application crashes. On every subsequent launch attempt, only the splash screen is briefly shown (about one second) before the application immediately closes.
However, Bambu Studio 01.10.01.50 is running stable!

### Expected results

It should start as usual.

### Project file & Debug log uploads

I cannot provide a project file because the application does not start at all. The issue is not related to a specific project.
The attached ZIP file contains four log files:
- The first log is from the initial launch after installation.
- The other three logs are from subsequent startup attempts.

[debug_Sat_Mar_29_08_50_56_9608.log.zip](https://github.com/user-attachments/files/19516616/debug_Sat_Mar_29_08_50_56_9608.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

lanewei120, 2025-03-31:
yes, currently bambu studio is not supported on windows arm64 platform

we are working on it

you need to use it under the offline mode(without the network plugin)

myshophosting, 2025-04-02:
Yes, but previous versions worked under the prism emulation. Somthing in the new release broke and now we cant use our printers.

Goodwine, 2025-04-02:
duplicate of https://github.com/bambulab/BambuStudio/issues/6014

lanewei120, 2025-04-07:
please check https://github.com/bambulab/BambuStudio/issues/6014#issuecomment-2782345560
for the current solution

we will fix it in next version

lanewei120, 2025-04-16:
fixed by v2.0.2.57(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57)

closed 2025-04-16 by lanewei120