bambulab/BambuStudio #6296: sketchup obj files crash this version of bamboo . 2016 sketchup????

p0ppy1976, 2025-03-29:
### Bambu Studio Version

win-v02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows

### Printer

a1

### How to reproduce

get obj file from 2016 slketchup

### Actual results

crashed bamboo

### Expected results

it does not work

### Project file & Debug log uploads

obj to bamboo

### Checklist of files to include

- [ ] Log file
- [x] Project file

p0ppy1976, 2025-03-29:
it did work in prev version ?

Haidiye00, 2025-03-29:
Hello, please provide the file and specific instructions. Thank you

p0ppy1976, 2025-03-29:
Hi 
obj file , these were opening fine , from sketchup 2016?

Regards

Simon 

---- On Sat, 29 Mar 2025 11:27:33 +0000 Haidiye00 ***@***.***> wrote ---

Haidiye00 left a comment https://github.com/bambulab/BambuStudio/issues/6296#issuecomment-2763307917

Hello, please provide the file and specific instructions. Thank you

Haidiye00, 2025-03-30:
Hello, so is there a problem now? If there is a problem, please provide the obj file

closed 2025-04-16 by Haidiye00