bambulab/BambuStudio #6297: Bambu Studio forgets the custom settings in AMS

vreichl, 2025-03-29:
### Bambu Studio Version

2.0.0.9.5

### Where is the application from?

Bambu Lab Official website

### OS version

MacOs Sonoma

### Additional system information

ImacPro 10 Cores 32gb ram

### Printer

X1C

### How to reproduce

I assign a user created material to a slot in AMS. After changing to another tab (like prepare) and changing back to the device page, the ams-slot doesn't show the assigned material anymore, but rather shows "?".

Same after restart of BambuStudio. No big deal assigning it again, but with time, it starts to be really annoying...

### Actual results

Bambu Studio "forgets" the assignment of the material in the AMS-Slot-Pictogram

### Expected results

Assigned material stays saved in the ams-pictogram until changing.

### Project file & Debug log uploads

[Adapter.zip](https://github.com/user-attachments/files/19517085/Adapter.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-29:
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
Hello, we suggest that you update to the latest version.
Also, can you take a screenshot or provide a GIF to explain your question? Thank you

JohnSmithCYM, 2025-03-29:
Thanks for your feedback, we want you to try this to test if it is a problem related to Studio: Can you just do the same action on the X1C screen(switch filament) ? 

Thanks for your feedback again.

vreichl, 2025-03-29:
I now updated BambuStudio to the latest version. Actually a print is running. After that print, I will check immediately.

closed 2025-04-16 by Haidiye00