bambulab/BambuStudio #6299: Slicing Result Color Scheme Default

eyescovered, 2025-03-29:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

P1S

### How to reproduce

1. Slice an object
2. Look at the Slicing Result Summary

### Actual results

By default it keeps going to Line Type, but I would prefer it to go to Filament Summary

[image]

### Expected results

It should be the Filament Summary

[image]

### Project file & Debug log uploads

any

### Checklist of files to include

- [ ] Log file
- [ ] Project file

HSMS-Rosmalen, 2025-06-12:
Fixed, hmm, I would expect to keep the last setting as a default, no other people will comment to have Line Type as the default option, including me.

eyescovered, 2025-06-12:
Makes sense, yeah. Should be whatever summary you used last

arekm, 2025-06-16:
[XunZhangBambu](https://github.com/XunZhangBambu) Please revert that (or make improved version where it remembers which type user used last)

"Line Type" is much more interesting than filament summary.

eremem, 2025-06-23:
> I recently updated and noticed this change as irritating and frustrating. I preferred "Line Type" as default. It makes much more sense to have it as default to make spotting issues at a glance easy again. At least add a option to change what it defaults to in the preferences menu. A simplyfied Filament Summary already gets displayed at the bottom of Line Type

I second that. Additionally with dark mode and black/dark filament the 3D view is literally useless when Color Scheme is Filament.

wildwoodashes, 2025-07-14:
Agreed. This is a huge oversight to not allow the user set the default. As everyone else has said, the 'filament' line type is nearly useless. It's fine if that's the out of the box (even though it's useless), but not giving us an option to set our own default is super annoying.

BenRinthLabs, 2025-07-28:
Agreed, please revert this or add a way to set the default