bambulab/BambuStudio #6300: Manually rotating the cutting plane does not reset to XYZ

coat64, 2025-03-29:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

X1C

### How to reproduce

Open a model and try to establish a cutting plane by rotating it over multiple planes.

### Actual results

When manually rotating the cutting-plane over multiple axis the XYZ axis rotates with it. This results in counterintuitive  results and makes it almost impossible to position a correct cutting-plane. When adjusting with one of the grips 1,2 or 3 rotating values will be altered.
Try to cut of the "arm" of the model I added.

[image]

### Expected results

The planes resets

[testcuttingplane.zip](https://github.com/user-attachments/files/19521287/testcuttingplane.zip)

 themself to the main XYZ system. Just like it does when using the normal rotating tool and just as it did in version 1.x.

### Project file & Debug log uploads

[testcuttingplane.zip](https://github.com/user-attachments/files/19521291/testcuttingplane.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

deletedalien, 2025-03-30:
I also noticed tgis bug and its super annoying especially considering this was completely different on older versions, it eas much more intuitive

guanyun-gudujian, 2025-03-31:
will be fixed next version

guanyun-gudujian, 2025-04-16:
@coat64  @deletedalien 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

deletedalien, 2025-04-17:
> [@coat64](https://github.com/coat64) [@deletedalien](https://github.com/deletedalien) fixed pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

yeah... no this didn't work.....

[image]

guanyun-gudujian, 2025-04-18:
> > [@coat64](https://github.com/coat64) [@deletedalien](https://github.com/deletedalien) fixed pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe
> 
> yeah... no this didn't work.....
> 
> [image]

sorry, it's an old issue, not an regression
will fix it
thanks

deletedalien, 2025-04-18:
aww man....
i for sure thought it was going to get fixed.....

deletedalien, 2025-04-30:
Is there any update on this??