bambulab/BambuStudio #6301: Failed to install Network Plug-in

Jasper-Ott, 2025-03-29:
### Bambu Studio Version

Bambu_Studio_mac-v02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia V 15.3.2

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Start up App
2. Request to install Network Plug-in to connect with devices
3. Agree and start install process

### Actual results

Gives a message that the install failed and says all VPNs need to be turned off.
Apple private relay is turned off and no other VPNs are installed on this device.

[image]

### Expected results

A correct install and connection with devices on the network.

### Project file & Debug log uploads

Bambu Studio software.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2025-03-29:
Have you tried to reboot your  Mac or restart your Mac? Some issue shows that can work.

JeffGoldner, 2025-03-29:
I filed this bug a day before. Restarting Bambu Studio fixed it, no need to reboot.

closed 2025-03-29 by Jasper-Ott