bambulab/BambuStudio #6302: number pad enter button

nochotwo, 2025-03-29:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

AMD Ryzen 9 6900HX with Radeon Graphics           3.30 GHz
32.0 GB

### Printer

Bambu abs A1/P1P

### How to reproduce

1. go to bambu studio.
2. open any project
3. resize or type in position
4. hit enter on number pad
5. watch nothing happen
6. hit enter button on main keyboard.
7. wonder why bambu removed that function

### Actual results

the enter button on the number pad does not work as an input

### Expected results

it should work as an input as in the versions before 2.x

### Project file & Debug log uploads

[debug_Sat_Mar_29_06_37_01_17892.log.zip](https://github.com/user-attachments/files/19521437/debug_Sat_Mar_29_06_37_01_17892.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

guanyun-gudujian, 2025-03-31:
will be fixed next version

guanyun-gudujian, 2025-04-16:
@nochotwo   @coat64 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe