bambulab/BambuStudio #6304: Support issues

Kin1024, 2025-03-29:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

WIN 10

### Printer

P1S  H2D  A1

### How to reproduce

Just slice it

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[support.zip](https://github.com/user-attachments/files/19522023/support.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

NothFie, 2025-03-31:
@Kin1024 Thanks for your report. We are aware of the issue and are actively working to fix the bug, but we still need some time to test it. We sincerely apologize for the inconvenience caused. and for now please manually apply support to the affected area.

NothFie, 2025-04-01:
This issue will be fixed in next version, sorry again for the inconvenience

NothFie, 2025-04-30:
The issue is fixed

closed 2025-04-30 by NothFie