bambulab/BambuStudio #6306: slicing not good leaves gaps between the lines

Mario-vminion, 2025-03-29:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macos

### Printer

bambu x1c

### How to reproduce

sliced a file with multi color
and had the gap ...

### Actual results

i sliced a file and had the line with 0.4 
there are  a lot of gaps between the line .. i changes the line with notting changes still the cap ...

[image]

### Expected results

that you see the gaps between the line of the different colours

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mario-vminion, 2025-03-30:
version 1 is oke  with this it started in version 2

BoBoYu-1, 2025-03-31:
Thank you for your feedback. Could you please provide the 3mf file?