bambulab/BambuStudio #6307: App crashes when opening a model from MakerWorld

mexusbg, 2025-03-29:
### Bambu Studio Version

v02.00.01.50-20250328181949

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Opening a model from MakerWorld crashes the app. When opening it again it crashes with unhandled error!

### Printer

P1S

### How to reproduce

1. Go to maker world
2. Try to open a model in BambuStudio

### Actual results

App crashed

### Expected results

App to open the model

### Project file & Debug log uploads

Not available, reverted to V1

If required will reinstall and provide logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-31:
Thank you for your feedback. Please provide the log file and DMP file, and also provide the makeworld model connection so that we can reproduce the problem and solve it.
The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

walterwongbbl, 2025-04-16:
Hi. @mexusbg  Please provide some useful information, such as the logs from Studio.
Or use our latest version.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57