bambulab/BambuStudio #6311: Bambu Studio looses connection to cloud if idle for too long.

gary-wlg, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.1

### Additional system information

This happened on the previous version of Bambu Studio as well.

I typically have several Bambu Studio windows open most of the day, but if I leave them idle for a couple hours and come back to them, I am unable to connect to any of my printers, I just get the error message "Failed to connect to the server" on all my open windows.  I am forced to close all windows, and re-open in order to re-establish communication with the cloud again.
 
It would be really nice if the application just re-connected to the cloud when it detects that it has lost the connection, it is a real pain to have to close all my windows and restart the application every time this happens.

### Printer

X1, P1, A1

### How to reproduce

Open a few instances of Bambu Studio... go have a coffee or two, then come back and try to connect to a printer.

### Actual results

Forced to close all Bambu Studio instances and re-open them to connect to cloud.

[image]

### Expected results

I would expect that the application just reconnects on its own when it looses connection, or provides a button to re-connect.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ddcr13, 2025-04-18:
I have the exact same problem here

monteua, 2025-04-20:
Same issue on the latest version of BambuStudio 2.0.2.57 and Mac OS 15.3.2

walterwongbbl, 2025-05-07:
@XinZhangBBL  用户版本是修复过问题之后的嘛。

XinZhangBBL, 2025-05-07:
@walterwongbbl  没有，MAC自动重连是4月份支持的。

XinZhangBBL, 2025-05-08:
@gary-wlg Halo. The new version has been released, please check if your problem has been solved. https://github.com/bambulab/BambuStudio/releases

[image]

XinZhangBBL, 2025-06-12:
@Dave-ToL Thank you for your feedback. We are looking in the issue. To get more information, could you reproduce the problem to offer log and video/screenshots for us. you can refer tohttps://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log