bambulab/BambuStudio #6312: Scaling crashes and is taking much longer than it used to

a-dam-berger, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: 11th Gen i9-11900H 2.5 GHz
RAM: 64GB
GPU: RTX 3080 Laptop

### Printer

A1 Combo

### How to reproduce

Using a multi-color model, if you scale the model up or down, it doesn't apply if you don't hit enter afterwards, and it crashes/lags when scaling if there is more than 1 copy on the print bed. Only a single plate is required. 4 color model. I have a commercial license for the file I am using, which is not publicly on MakerWorld, so I can't share the actual file, but I have had this issue with pretty much every model when trying to scale it since the new H2D updates.

### Actual results

It lags and takes a lot longer to resize the model than previously. At least 15 seconds, but sometimes minutes.

### Expected results

It should scale up within a few seconds as it used to do. Also, you previously didn't have to hit enter after you changed the scale for it to apply.

### Project file & Debug log uploads

[MMMini_Bunny_Multicolor_v01_For BugReport.zip](https://github.com/user-attachments/files/19523735/MMMini_Bunny_Multicolor_v01_For.BugReport.zip)
[log.zip](https://github.com/user-attachments/files/19523734/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-03-31:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

a-dam-berger, 2025-04-01:
Today I am not seeing the lagging that I was previously, but I did install the new release of Bambu Studio yesterday.

That said, for some reason, the 10 Key ENTER still doesn't apply the scaling. Previously this did work. Can you fix it so both Enter Keys can be used?

I commonly change numeric values like the scale with the 10 Key.

Haidiye00, 2025-04-02:
Hello, we expect to fix it around mid April. At present, you can use the enter button in English first, or after entering, click on the blank space of the imgui interface with the mouse