bambulab/BambuStudio #6318: Bring back being able to change dimansions on Prime Tower

CDub0512, 2025-03-30:
Itsw ridiculous how you took away the able to change the dimensions on the Prime Tower. Bring it back..

coat64, 2025-03-30:
If you disable "rib wall" you can change the width.

[image]

coat64, 2025-03-30:
> You can not lower the total volume of the prime tower though It still wastes just as much filament

Yes you seem to be right about that. They set it to a more or less fixed amount it seems. Changing the width alters the length.

CDub0512, 2025-03-30:
Yeah it doesn't make it like it used to be.

OGSRLibrary, 2025-04-02:
Without Prime Volume I've had to roll back to studio 1.10. 
Some models are too large to print without having fine control over the Prime Tower dimensions and placement. Disabling Prime Tower completely lowers the print quality too much, particularly with regard to capturing the small amount of ooze after the filament change. Adjusting Width and Infill Gap are not enough.
Please bring this control back, many of my old plates no longer slice without it.