bambulab/BambuStudio #6319: Can't add SUNLU PETG Filament

Cham-HL, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Additional system information

I can add SUNLU other filament typr as system filament but not SUNLU PETG, even I select SUNLU PETG on the list but not shown on project filemant pull doen menu.

### Printer

Bambu A1 mini

### How to reproduce

1. select "add/remove filaments" from pull down menu below "Project filament"
2. tick "SUNLU PETG" and press confirm
3. no SUNLU PETG shown on pull down menu before "add/remove filaments"
4. but ok if I select other SUNLU filaments

[image]

[image]

 

### Actual results

[image]

### Expected results

should have a SUNLU PETG option to select in the pull down menu

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DirkFlorin, 2025-03-30:
I have been having the same issues with filament profiles. It looks to me, that some profiles are just made for a specific printer. For example they appeared only for my X1C, but not for the H2D and when I was taking a closer look into the filaments profiles json files, they had only the X1/C/E, P1P/S as "compatible" printers listed. I manually created a profile and copied the values over it or exported the filament profile and changed it manually.

Maybe that's the case here as well.

Cham-HL, 2025-03-30:
@DirkFlorin, Thanks for your valuable information. As a greenhorn,  I'll try using generic PETG first and see the printout result.

DirkFlorin, 2025-04-02:
@Cham-HL You are more than welcome. We all started with this at some point. 

But honestly, the Generic profiles sometimes working pretty good. Usually, I change the nozzle or bed temps, based on the recommendation of the filament manufacturer or when I encounter printing issues or degradation of quality. And don't worry, everything will come with the time and experience you gain from failing.

XunZhangBambu, 2025-04-09:
@Cham-HL  It's a preset from user's pull request. We didn't test the filaments offically and all valiable presets for sunlu came from that PR

rdunin, 2025-04-12:
Hi, I have a similar problem. When @XunZhangBambu mentioned that Sunlu PR added a pull request, I tried to find what was added. I couldn’t find it by searching for:
repo:bambulab/BambuStudio sunlu petg

It looks like they added PETG presets only for 0.2mm and 0.8mm nozzles, but the 0.4mm nozzle PETG settings are missing.

king1979, 2025-04-23:
#6017