bambulab/BambuStudio #6320: Scarf seam not working in BS 2.0.1.50

frasier2k, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Enable Scarf Seam in filament settings.
Slice Model.

### Actual results

With BS 2.0.1.50 scarf seam feature is not working for me. I tried with prev version 1.10.2.76 and there it is working fine.
Do I miss something?

See attached screenshots.

[image]
[image]

### Expected results

Scarf seam should working as in prev version.

### Project file & Debug log uploads

[scarf seam test.zip](https://github.com/user-attachments/files/19525058/scarf.seam.test.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

QingZhangBambu, 2025-04-01:
i will check this.

Ben00793, 2025-04-02:
Same issue here, with BS 2.0.1.50. Scarf not working.

QingZhangBambu, 2025-04-03:
I have identified the issue and am working on fixing it.