bambulab/BambuStudio #6328: Lay on face - Convex Hull not properly orienting parts

nem3, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C, A1, A1 mini

### How to reproduce

Load a included model and use the lay on face feature to select the front side.

[Windshield Frame.zip](https://github.com/user-attachments/files/19526085/Windshield.Frame.zip)

### Actual results

Part is slightly tilted up from bed lengthwise. Resulted in incorrect print quality and unneeded supports. Part orientation can sometimes be corrected manually by zeroing out the extraneous values in X, Y and Z.

[image]

### Expected results

Expect parts to be laid on flat face.

[image]

### Project file & Debug log uploads

[Windshield Frame Project.zip](https://github.com/user-attachments/files/19526122/Windshield.Frame.Project.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

tangketan, 2025-04-23:
Thanks for reporting the issue. It's a very interesting case. We'll look into it.
For now you can use "auto orient" funciton which gives correct result in this case.

Haidiye00, 2025-04-23:
[image]
[image]
Hello, after our analysis, this is not a bug.
It is recommended to use the automatic orientation function of the disc to ensure that the bottom plane of the object is almost aligned with the disc plane.
And lay on face does not necessarily mean that the bottom plane of the object is almost identical to the plane of the disk.
There is a certain difference in algorithm between the two functions.
We also provide the normal vectors of the models obtained from the two specific functions above.
[image]