bambulab/BambuStudio #6330: Copilot PC with Qualcomm chip crashing on V2

qdewolf, 2025-03-30:
### Bambu Studio Version

02.00.00.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Snapdragon X Elite with 32GB RAM, 1TB storage

### Printer

Bambu Labs X1c

### How to reproduce

1. Install Studio
2. when app launches it asks to install network plugin
3. click OK

### Actual results

when i install the new studio version 2 it installs and runs fine if I don’t install the network plugin. if I install the network plugin then it crashes as soon as the plugin is finished installing and when I run it, I see the splash screen come and go but no app. No app in task manager.

### Expected results

app runs

### Project file & Debug log uploads

does not get far enough to create debug

### Checklist of files to include

- [ ] Log file
- [ ] Project file

voseldop, 2025-03-31:
I have Surface Pro with Windows on ARM and the same issue with V2.0.1.50.

lanewei120, 2025-03-31:
yes, currently bambu studio is not supported on windows arm64 platform

we are working on it

you need to use it under the offline mode(without the network plugin)

Goodwine, 2025-04-02:
duplicate: https://github.com/bambulab/BambuStudio/issues/6014

lanewei120, 2025-04-07:
Hello

please download this file
[win_02.00.01.03.zip](https://github.com/user-attachments/files/19627997/win_02.00.01.03.zip)
and unzip it into your plugin directory of bambu studio
(C:\Users\user.name\AppData\Roaming\BambuStudio\plugins)

[image]

then restart bambu studio, it should be ok

please have a try with this way currently
we will fix it in the next version

lanewei120, 2025-04-16:
fixed by v2.0.2.57(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57)

closed 2025-04-16 by lanewei120